LSIR/gsn #50: GSN Build issue

varks, 2015-05-21:
Hi All,

I have been trying to build GSN from the source code and every time it fails with the below error. I am facing this issue since 10 days. It used to work fine before that. It will be helpful if someone can help me resolve this issue.

Steps to reproduce the issue:  
1. On a fresh box/VM, Install java, ant and other packages required for GSN. 
2. git clone https://github.com/LSIR/gsn.git
3. cd gsn
4. ant
5. ant gsn

Build log:

[artifact:dependencies]   org.apache.geronimo.specs:geronimo-stax-api_1.0_spec:jar:1.0.1
[artifact:dependencies]
[artifact:dependencies] from the specified remote repositories:
[artifact:dependencies]   apache.snapshots (http://repository.apache.org/snapshots),
[artifact:dependencies]   central (http://repo1.maven.org/maven2),
[artifact:dependencies]   TypesafeRepository (http://repo.typesafe.com/typesafe/releases/),
[artifact:dependencies]   LSIR (http://osper.epfl.ch:8081/artifactory/gsn-release/)
[artifact:dependencies]
[artifact:dependencies] Path to dependency:
[artifact:dependencies]         1) gsn:gsn:jar:1.1.5
[artifact:dependencies]         2) org.apache.axis2:axis2-adb:jar:1.5.5
[artifact:dependencies]         3) org.apache.axis2:axis2-kernel:jar:1.5.5
[artifact:dependencies]         4) org.apache.ws.commons.axiom:axiom-api:jar:1.2.11
[artifact:dependencies]
[artifact:dependencies]
[artifact:dependencies] Not a v4.0.0 POM. for project org.apache.geronimo.genesis.config:config at /root/.m2/repository/org/apache/geronimo/genesis/config/config/1.1/config-1.1.pom

BUILD FAILED
/home/ec2-user/gsn/build.xml:480: Unable to resolve artifact: Unable to get dependency information: Unable to read the metadata file for artifact 'org.apache.geronimo.specs:geronimo-stax-api_1.0_spec:jar': Cannot find parent: org.apache.geronimo.genesis.config:config for project: null:project-config:pom:1.1 for project null:project-config:pom:1.1
  org.apache.geronimo.specs:geronimo-stax-api_1.0_spec:jar:1.0.1

from the specified remote repositories:
  apache.snapshots (http://repository.apache.org/snapshots),
  central (http://repo1.maven.org/maven2),
  TypesafeRepository (http://repo.typesafe.com/typesafe/releases/),
  LSIR (http://osper.epfl.ch:8081/artifactory/gsn-release/)

Path to dependency:
        1) gsn:gsn:jar:1.1.5
        2) org.apache.axis2:axis2-adb:jar:1.5.5
        3) org.apache.axis2:axis2-kernel:jar:1.5.5
        4) org.apache.ws.commons.axiom:axiom-api:jar:1.2.11

Total time: 1 second

  . 

Thanks,
Varun

ebiiii, 2015-05-21:
Hello Varun,
I just re-run the travis CI to be sure there wasn't any incompatible changes in the dependencies, but it seems that it is perfectly able to build it (https://travis-ci.org/LSIR/gsn).
what is the content of your file /root/.m2/repository/org/apache/geronimo/genesis/config/config/1.1/config-1.1.pom? It seems it has some metadata issues...
Best,
Julien

varks, 2015-05-21:
Hi Julien,

I see this issue only when doing a fresh install on a new machine, which
means some of remote repositories might have been moved to a different link
and those have not been updated.

The content of the file you asked  is as below:

<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>302 Found</title>
</head><body>
<h1>Found</h1>
<p>The document has moved <a href="
https://www.codehaus.org/termination-503.html">here</a>.</p>
<hr>
<address>Apache/2.2.3 (Red Hat) Server at repository.codehaus.org Port
80</address>
</body></html>

Thanks,
Varun

jpcik, 2015-05-21:
It seems that codehaus repo is unavailable since a few days. Taht genesios config pom was fetched from there.
Putting this in the pom.xml, before the axis dependency should fix it:
		<dependency>
			<groupId>org.apache.geronimo.genesis.config</groupId>
			<artifactId>config</artifactId>
			<version>1.1</version>
			<type>pom</type>
		</dependency>

will check and see if it gets fixed

jpcik, 2015-05-21:
the issue should be reproducible if cleaning the maven cache: e.g. contents of the .m2/repository folder.

varks, 2015-05-21:
I tried what you suggested. It still does not fix it.

[artifact:dependencies] Transferring 42K from TypesafeRepository
[artifact:dependencies] An error has occurred while processing the Maven
artifact tasks.
[artifact:dependencies]  Diagnosis:
[artifact:dependencies]
[artifact:dependencies] Unable to resolve artifact: Missing:
[artifact:dependencies] ----------
[artifact:dependencies] 1) org.apache.geronimo.genesis.config:config:jar:1.1
[artifact:dependencies]
[artifact:dependencies]   Try downloading the file manually from the
project website.
[artifact:dependencies]
[artifact:dependencies]   Then, install it using the command:
[artifact:dependencies]       mvn install:install-file
-DgroupId=org.apache.geronimo.genesis.config -DartifactId=config
-Dversion=1.1 -Dpackaging=jar -Dfile=/path/to/file
[artifact:dependencies]
[artifact:dependencies]   Alternatively, if you host your own repository
you can deploy the file there:
[artifact:dependencies]       mvn deploy:deploy-file
-DgroupId=org.apache.geronimo.genesis.config -DartifactId=config
-Dversion=1.1 -Dpackaging=jar -Dfile=/path/to/file -Durl=[url]
-DrepositoryId=[id]
[artifact:dependencies]
[artifact:dependencies]   Path to dependency:
[artifact:dependencies]         1) gsn:gsn:jar:1.1.5
[artifact:dependencies]         2)
org.apache.geronimo.genesis.config:config:jar:1.1
[artifact:dependencies]
[artifact:dependencies] ----------
[artifact:dependencies] 1 required artifact is missing.

Thanks,
Varun

jpcik, 2015-05-21:
did you clean your maven local repo? you probably have the invalid config pom in your .m2

varks, 2015-05-21:
Yes. I cleaned the maven repo and now it no longer shows dependencies error.

Thank You Jean

varks, 2015-05-21:
Hi Jean,

But I still hit other errors like these:

[artifact:dependencies] Its dependencies (if any) will NOT be available to
the current build.

init:

setup:
    [mkdir] Created dir: /home/ec2-user/gsn/target/classes
    [mkdir] Created dir: /home/ec2-user/gsn/logs
    [mkdir] Created dir: /home/ec2-user/gsn/target/reports/todos
    [mkdir] Created dir: /home/ec2-user/gsn/target/reports/javadoc
    [mkdir] Created dir: /home/ec2-user/gsn/target/reports/junit
     [echo] 1.6

build:
    [javac] Compiling 481 source files to /home/ec2-user/gsn/target/classes
    [javac] error: error reading
/root/.m2/repository/com/github/stephenc/findbugs/findbugs-annotations/1.3.9-1/findbugs-annotations-1.3.9-1.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/google/code/findbugs/jsr305/1.3.9/jsr305-1.3.9.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/google/guava/guava/12.0.1/guava-12.0.1.jar; error
in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/google/inject/extensions/guice-servlet/3.0/guice-servlet-3.0.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/google/inject/guice/3.0/guice-3.0.jar; error in
opening zip file
    [javac] error: error reading
/root/.m2/repository/com/google/protobuf/protobuf-java/2.5.0/protobuf-java-2.5.0.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/jcraft/jsch/0.1.42/jsch-0.1.42.jar; error in
opening zip file
    [javac] error: error reading
/root/.m2/repository/com/sun/jersey/contribs/jersey-guice/1.9/jersey-guice-1.9.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/sun/jersey/jersey-core/1.9/jersey-core-1.9.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/sun/jersey/jersey-json/1.9/jersey-json-1.9.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/sun/jersey/jersey-server/1.9/jersey-server-1.9.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/com/thoughtworks/paranamer/paranamer/2.3/paranamer-2.3.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/io/netty/netty-all/4.0.19.Final/netty-all-4.0.19.Final.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/javax/servlet/servlet-api/2.5/servlet-api-2.5.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/javax/xml/stream/stax-api/1.0-2/stax-api-1.0-2.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/org/codehaus/jackson/jackson-core-asl/1.8.8/jackson-core-asl-1.8.8.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/org/codehaus/jackson/jackson-mapper-asl/1.8.8/jackson-mapper-asl-1.8.8.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/org/htrace/htrace-core/3.0.4/htrace-core-3.0.4.jar;
error in opening zip file
    [javac] error: error reading
/root/.m2/repository/org/tukaani/xz/1.0/xz-1.0.jar; error in opening zip
file
    [javac] error: error reading
/root/.m2/repository/org/xerial/snappy/snappy-java/
1.0.4.1/snappy-java-1.0.4.1.jar; error in opening zip file
     [copy] Copying 2 files to /home/ec2-user/gsn/target/classes

BUILD SUCCESSFUL
Total time: 4 seconds

Any suggestions ?

jpcik, 2015-05-21:
I just commited this change to the master. you may clean all the .m2 stuff, pull from master and do ant build.
it's strange that it shows those errors. You can also try mvn compile, to see if it works. travis seems to build without problems

varks, 2015-05-21:
I cleaned the m2 cache and pulled your latest changes. I still face the
same issue.

Coincidently, all the erroneous pom files have HTTP 302 message. and are
from codehoas repo . Is there are any other repo we can rely on. A couple
of pom's like the protobufs, have checksum issues and thus the
corresponding jars are not being fetched.

jpcik, 2015-05-21:
strange issue. as it compiles with maven, I can suggest you to clean cache (again), do mvn compile, and only then do ant build.
mvn should download the poms and jars from mvn central repo and not form codehaus.

closed 2015-07-28 by ebiiii

shebajijugeorge, 2016-01-20:
While i installed gsn in ubuntu14.04 i got the error as follows
gsn-master$ ant build
Buildfile: /home/jikku/gsn-master/build.xml
[artifact:dependencies] An error has occurred while processing the Maven artifact tasks.
[artifact:dependencies]  Diagnosis:
[artifact:dependencies] 
[artifact:dependencies] Unable to resolve artifact: Unable to get dependency information: Unable to read the metadata file for artifact 'org.htrace:htrace-core:jar': Cannot find parent: org.htrace:htrace for project: null:htrace-core:jar:null for project null:htrace-core:jar:null
[artifact:dependencies]   org.htrace:htrace-core:jar:3.0.4
[artifact:dependencies] 
[artifact:dependencies] from the specified remote repositories:
[artifact:dependencies]   apache.snapshots (http://repository.apache.org/snapshots),
[artifact:dependencies]   central (http://repo1.maven.org/maven2),
[artifact:dependencies]   codehaus (http://repository.codehaus.org/),
[artifact:dependencies]   java.net (http://download.java.net/maven/2/),
[artifact:dependencies]   ghelmling.testing (http://people.apache.org/~garyh/mvn/),
[artifact:dependencies]   repository.jboss.org (http://repository.jboss.org/nexus/content/groups/public-jboss/),
[artifact:dependencies]   TypesafeRepository (http://repo.typesafe.com/typesafe/releases/),
[artifact:dependencies]   LSIR (http://osper.epfl.ch:8081/artifactory/gsn-release/),
[artifact:dependencies]   apache release (https://repository.apache.org/content/repositories/releases/)
[artifact:dependencies] 
[artifact:dependencies] Path to dependency: 
[artifact:dependencies] 	1) gsn:gsn:jar:1.1.5
[artifact:dependencies] 	2) org.apache.hbase:hbase-common:jar:0.99.0
[artifact:dependencies] 
[artifact:dependencies] 
[artifact:dependencies] Not a v4.0.0 POM. for project org.htrace:htrace at /home/jikku/.m2/repository/org/htrace/htrace/3.0.4/htrace-3.0.4.pom

BUILD FAILED
/home/jikku/gsn-master/build.xml:481: Unable to resolve artifact: Unable to get dependency information: Unable to read the metadata file for artifact 'org.htrace:htrace-core:jar': Cannot find parent: org.htrace:htrace for project: null:htrace-core:jar:null for project null:htrace-core:jar:null
  org.htrace:htrace-core:jar:3.0.4

from the specified remote repositories:
  apache.snapshots (http://repository.apache.org/snapshots),
  central (http://repo1.maven.org/maven2),
  codehaus (http://repository.codehaus.org/),
  java.net (http://download.java.net/maven/2/),
  ghelmling.testing (http://people.apache.org/~garyh/mvn/),
  repository.jboss.org (http://repository.jboss.org/nexus/content/groups/public-jboss/),
  TypesafeRepository (http://repo.typesafe.com/typesafe/releases/),
  LSIR (http://osper.epfl.ch:8081/artifactory/gsn-release/),
  apache release (https://repository.apache.org/content/repositories/releases/)

Path to dependency: 
	1) gsn:gsn:jar:1.1.5
	2) org.apache.hbase:hbase-common:jar:0.99.0

anyone please suggest a solution to resolve it

ebiiii, 2016-01-20:
Hi Sheba,
It seems that Maven failed at retrieving the dependency 'org.htrace:htrace-core:jar', maybe just re-running it could solve the issue. Or clearing the maven cache, in case it cached a corrupted file.
Let us know if it solves your issue.

shebajijugeorge, 2016-01-20:
how to clear MAVEN cache ? since i tried the below command to clean cache i got an error
 mvn clean install
Segmentation fault (core dumped)

ebiiii, 2016-01-20:
Sorry, I was talking about the downloaded artifacts that are stored in the .m2 folder in your home directory. In your case I would delete the folder ~/.m2/repository/org/htrace/htrace-core 
and then re-launch ant build.