LSIR/gsn #51: GSN Web UI latitude/longitude issue

kjorg50, 2015-05-28:
I noticed that if I create a CSV virtual sensor and use the field names "latitude" and "longitude," then on the web page the actual `<addressing>` values for latitude and longitude get shown next to the data values. For example, when the virtual sensor is first loaded
[image]

But after the first auto-refresh occurs it shows this
[image]

I used 0.0, 0.0 as my latitude/longitude in the `<addressing>` section of the virtual sensor. All of the data is recorded in the database properly, so it's just an issue in the view. Is there a simple solution for this? Thanks!

ebiiii, 2015-06-02:
Hello,

It looks indeed like a bug in the javascript of the UI. At https://github.com/LSIR/gsn/blob/master/src/main/webapp/js/gsn.js#L945 , when getting the list of fields, there is no distinction between actual dynamic fields and predicates. The XPath could be made more selective. 
As we are working on a completely new interface, we won't have much time for fixing the old one. But this seems to be an easy one.

Thanks!

kjorg50, 2015-06-03:
Okay, looking forward to the new UI!

ebiiii, 2015-07-28:
As the new UI is getting a bit late, this (d317c1dcae8db562bda0bd24f7d459c8beb8a77c) should solve the issue.

closed 2015-07-28 by ebiiii