LSIR/gsn #57: bash: ./gsn-start.sh: /bin/bash^M: bad interpreter: No such file or directory

iseitani, 2015-09-24:
I have the following problem with the Unix version : 
bash: ./gsn-start.sh: /bin/bash^M: bad interpreter: No such file or directory
Thank you!

ebiiii, 2015-09-24:
Hello,
Thanks for the bug report!
it seems it is a problem with the line endings... (the file was probably edited on Windows).
Are you using the version from the installer ?

As a quick solution you can open the file gsn-start.sh and remove all the "^M" that would appear at the end of each line. We will correct this and update the installer as soon as possible.

Thanks

ebiiii, 2016-02-05:
Solved in b4487aff9f65c93d3a238e315e9b602f73ffb7c5. The new installer is available in the release 1.1.8.

closed 2016-02-05 by ebiiii