LSIR/gsn #58: implement a CoAP wrapper

ebiiii, 2015-09-24:
can use this java library: https://github.com/okleine/nCoAP for example

m6461, 2016-02-01:
Hey, have you already implemented it? If not, may I try? I am new here, so sorry if the question is naive.

ebiiii, 2016-02-02:
Hello, no we didn't started yet with this one. So feel free to give it a try. You can find some documentation on how to implement wrappers here: https://github.com/LSIR/gsn/wiki/how-to-write-a-new-wrapper . Let me know if you have questions.

m6461, 2016-02-02:
thanks. will get back to you if i face some problems.

iseitani, 2016-02-05:
@m6461  if you need I have the wrapper and virtual sensor for CoAP using Californium

m6461, 2016-02-05:
@iseitani that would be a great help. thanks a lot in advance. you can send in md.anowarul.abedin@ltu.se, or just let me know how I should collect.

iseitani, 2016-02-05:
You will need californium-core & element-connector libraries(you can find them on mvnrepository).
I had a few system crashes and didn't commit my code, I hope I will do this the following weeks.

[GSN-COAP.zip](https://github.com/LSIR/gsn/files/119202/GSN-COAP.zip)

m6461, 2016-02-05:
nice! if necessary, i will modify it and will try to improve. will keep you updated about it.

ebiiii, 2016-11-02:
Thanks @iseitani, the CoAP wrapper has been added to the defaults wrappers : 339dafe2c810e47fb762cb09a593fd6f8dfa12e4

Feel free to re-open the issue if you have any trouble using it.

closed 2016-11-02 by ebiiii