LSIR/gsn #69: Tiny-GSN Functionalities

m6461, 2016-02-18:
Hello,
What are the functionalities available in Tiny-GSN? Can it push data to XGSN? Can it sense data semantically? Read me file does not give all these details. Thanks.

ebiiii, 2016-02-22:
Hello,
the latest version of tinyGSN is able to collect data from any internal sensor and potentially external ones (using the same notion of wrapper as in GSN, one can easily implement the support for a new sensing device). It has a configurable stream processing, through virtual sensors very similar to the ones in GSN (porting the code should be trivial). It can subscribe to a GSN data stream (currently by polling the API, but some push mechanisms are being developed). It can publish its data to a GSN server (periodically, or whenever a wifi connection is available). It integrates a privacy protection library for obfuscating location data.
XGSN being a fork of an old version of GSN, it is not supported by tinyGSN. 
What do you mean by semantically? if you want to transform a stream of raw GPS coordinates into places, or accelerometer values into user activities you can easily implement a classifier (or any method you can find in the literature) inside a virtual sensor.
Maybe @jpcik can answer you more precisely on the semantic part...

m6461, 2016-02-24:
Thanks, ebiiii, it helps.

closed 2016-04-05 by ebiiii