LSIR/gsn #79: Error in input streams

mvishnuprathap, 2017-01-04:
Some virtual sensors in my gsn installation doesn't work due to error in input streams. Can anyone help me to solve this issues ?

ebiiii, 2017-01-05:
Dear mvishnuprathap, do you have some more concrete error message or logs? What is your setup (GSN version, list of virtual sensors, OS and version)? This could definitely help in identifying the root cause of these errors.

mvishnuprathap, 2017-01-05:
Dear ebiiii, thanks for your reply. the os version i'm using is ubuntu 14.04 LTS, 64 bit OS. The error occured when I tried to simulate the sensor named with the xml file " sensorscope2sample.xml ". I'm attaching the log of the error occured when I simulated
[gsn_sensorscopesample2_error.txt](https://github.com/LSIR/gsn/files/687740/gsn_sensorscopesample2_error.txt)

ebiiii, 2017-01-06:
I see you opened another issue for this. So I'll close this one.

closed 2017-01-06 by ebiiii