LSIR/gsn #83: GSN configuration error

AndreaLCN, 2017-07-28:
Hi all,

I'm trying to build GSN project but something get wrong.
I do not Know if I'm making some mistake building it. The procedure that produce the errors is: 

I use SBT. First step a build it with sbt command. (obviously in the correct directory - gsn directory).
Second step, I type project webui and then StartDjango.
Third step I typeproject core and then re-start. Here the first error. Attached file: project_core_error.
[image]

Forth step. I type project servicer and then run. Here an other error. attached files --> project_services_error_1, project_services_error_2, project_services_error_3.
[image]
[image]
[image].

Last step i try localhost:9000, and here a configuration error, maybe due to the others error. Attached file: configuration_error.
[image]

I would like to know if these errors are due to a wrong procedure that i'm doing to build the project or to something else.
Can somebody help me to fix them? 

thanks,
Andrea.

ebiiii, 2017-07-31:
Hi Andrea,

The first error, which could be actually a warning, is just because some configuration files are missing in the gsn-core project, and it is using the default parameters, which are just fine.

The second error is because of this configuration line: https://github.com/LSIR/gsn/blob/master/gsn-services/conf/application.conf#L15 . It is meant for when the application is packaged and installed, but for running it from the sources, you will have to change it to point to the gsn-core folder. 

I'll keep this issue open to remind me to change that next time I have some time to update it. :)

AndreaLCN, 2017-08-01:
Thank u Julien. If I'll have some other problem I'll contact you. Have a
nice day.

Saluti, A.Serra

https://github.com/LSIR/gsn/blob/master/gsn-services/conf/
application.conf#L15 . It is meant for when the application is packaged and

<https://github.com/LSIR/gsn/issues/83#issuecomment-318968911>, or mute the
thread

AndreaLCN, 2017-08-01:
Hi Julien, 

I did what you suggested but it doesn't work! I mean, I hope that I have correctly understood what you mean. So I change in tha applicantion.conf file the line:  
gsn.location="/usr/share/gsn-core/" with --> gsn.location="/home/ubuntu/Desktop/gsn-master/gsn-core/" 
Is it correct?

ebiiii, 2017-08-02:
oh, my bad. The "conf" folder is at the root in the repository. the path should be then:  gsn.location="/home/ubuntu/Desktop/gsn-master/"

AndreaLCN, 2017-08-02:
Well, you was right. Tha correct path was the last you wrote, but now an other problem occurs. I'm so sorry. 
[image]
Have you ever seen an error like that? How can I fix it? I saw that an error like that was just post in an other issues tab but without any answer. I'd really appreciate if you can still help me.

ebiiii, 2017-08-03:
This looks like an error of the gsn-services. The default database is an H2 file in /tmp/GsnAuthDb. It should be initialized and all tables created with the evolution scripts. Maybe deleting this file and restarting the services could work ?

AndreaLCN, 2017-08-03:
Yes, it looks like an error due to that file, but if I delete it the problem is not fixed! 
Have a look! it's a kind of problem due to play and to this database. And there is still this internal server error.
[image]
[image]
[image]

ebiiii, 2017-08-08:
I remember I already had this issue. The evolution scripts in gsn-services/conf/evolutions/default/
were automatically re-generated according to the database driver that was selected in the application.conf . But at some point they stopped being updated and the current version looks like the postgresql one. According to https://stackoverflow.com/questions/9964266/how-to-enable-ddl-generation-in-playframework-2-0 this is when I added the second file there. A quick hack could be to remove the 2.sql file, run the service, and add it back and restart the service. Let me know if it works...

AndreaLCN, 2017-08-08:
I tried but it does not work! But I was reading the error, it is a syntax error in SQL statement. Is it possible that this error is due to lowercase?

ebiiii, 2017-08-10:
Yes, the syntax is slightly different between the DB systems. In this case, you can also manually edit the file 1.sql to fix it. (I remember postgres had issue with names longer than 31 char, but I don't remember the issue with H2). Actually the file you have now in gsn-services/conf/evolutions/default/ is different from the one here https://github.com/LSIR/gsn/blob/master/gsn-services/conf/evolutions/default/1.sql
If you can figure out what is wrong, I'd be happy to update the repository or add it in the documentation ;-)

AndreaLCN, 2017-08-11:
Ok, I checked my 1.sql file and it is different from which one you linked. I tried to modify it but when I re-run the program the file 1.sql return the same as before it was modified. How can I effectively modify it? Do you want that I point you out the differences?

ebiiii, 2017-08-11:
I think if you remove the comments at the beginning of the file it would stop rewriting it. To be tested

AndreaLCN, 2017-08-11:
ok, I tried to fix these errors, I mean those in 1.sql file. Now it gives me an error in the 2.sql file. It is an error due to security_role. Maybe because security_role has 2 arguments instead of 3!? 
Have a look
[image]

ebiiii, 2017-08-11:
well the security_role table should be correct with 2 columns: https://github.com/LSIR/gsn/blob/master/gsn-services/conf/evolutions/default/1.sql#L75. I'm not really sure that the 1.sql really created everything then.

AndreaLCN, 2017-08-12:
> evolutions/default/1.sql#L75. I'm not really sure that the 1.sql really

Ok, it should be correct. But the error now is that this security_role is
not found! How is it possible? Does it depend on the 1.sql file?

Saluti, A.Serra

AndreaLCN, 2017-08-12:
Have a look.
[image]

ebiiii, 2017-08-14:
1.sql evolution does create the tables and 2.sql populates them with some initial users and clients. If a table is missing, it is most likely that 1.sql failed somewhere at creating the given table.
How does 1.sql looks like now that you fixed the issues with it? 
Is this error happening also when you delete the /tmp/GsnAuthDb?

AndreaLCN, 2017-08-14:
The last screen that I posted appears because I deleted the /tmp/GsnAuthDb, but it still doesn't find the security_role table.

ebiiii, 2017-08-16:
And which modifications did you do in 1.sql?

AndreaLCN, 2017-08-16:
I exactly copied this one: https://github.com/LSIR/gsn/blob/master/gsn-services/conf/evolutions/default/1.sql

AndreaLCN, 2017-08-19:
Hi Julien, 
do you have any news about that problem? But if you build the project in yout PC, does it work?

ebiiii, 2017-08-21:
Hi Andrea, sorry for the delay. So I managed to run it locally on my new laptop (configuring everything from scratch). For the evolution to work you have to edit the file 1.sql:
- remove the two first lines of comment that says that it was generated by ebean DDL
- around the lines 34 and 42, change "unique (group, data_source)" to "unique (group_id, data_source_id)"
- around lines 109 and 117 change "(user, data_source))" to "(user_id, data_source_id))"
Then remove the database file in /tmp and start the service. 

It seems that the ebean generator for the evolution has a bug when the column names are reserved words (which I agree is not supposed to happen to a well coded project :-P ). and when writing the unique constraints it doesn't append the "_id" to the identifiers.

Let me know if this solves the problem

AndreaLCN, 2017-08-21:
Don't worry if you answerd me late =p. Anyway it doesn't work because my 1.sql file was just correct with that modifications. I exactly copied the code in https://github.com/LSIR/gsn/blob/master/gsn-services/conf/evolutions/default/1.sql that it looks like correct. 
It's strange :(

ebiiii, 2017-08-23:
From the file in  https://github.com/LSIR/gsn/blob/master/gsn-services/conf/evolutions/default/1.sql you still need to remove the two first lines. Otherwise it will get overwritten by the ebean evolution.

AndreaLCN, 2017-08-23:
Yes, I tried also deleting them, but the error or better this unexpected exception still remain.

dmaiorca, 2017-08-30:
Dear all,
I also managed to make the application correctly run by performing the changes mentioned by Julien. 
I think that the key point here is that the line with the "ups" stuff must be kept in the file (like Julien said, only the first two should be deleted). Besides, it is crucial to delete the contents of the /tmp/ folder related to the dabatase before re-running the service. 

I am attaching the file that worked for me.
[1.zip](https://github.com/LSIR/gsn/files/1263235/1.zip)

Hope it solves the problem.
Best,
Davide

AndreaLCN, 2017-09-10:
Hi Julien, 
thanks to dmairoca I fixed all the problems to build the project. Now when I try to log in in localhost with root@localhost as username and changeme as password it appears this sentence: "This client is not registered for accessing GSN, please contact the administrator if you need to add it." What exactly do i have to do to log in? Thank you.

AndreaLCN, 2017-10-07:
Hi Julien, 

sorry for my infinite questions XD 
When I go, after I launched Django in localhost:8000, the page it apears like the following image: 
[image]
 Is it correct that it apperas the white blank or it has to appear the world map where I should find sensors?
Thanks a lot and have a nice day.

ebiiii, 2017-10-09:
Hi Andrea,
It looks like the javascript and css libraries are not loaded. How did you start the django app? Maybe the "bower install" step didn't complete successfully ?

AndreaLCN, 2017-10-09:
I started Django after launch sbt. Once lauched i chage directory in the
project webui and there I launch startDjango. Is it correct? It doesn't
seem bower has problems but i'm going to check ;)

Saluti, A.Serra